nordicsemi/IOS-nRF-Toolbox #1: DFU max speed

alexisduque, 2015-05-30:
Hi,
What's the max throughput we can achieve with iOS (4S) during DFU ? What are reliable connection interval, packet size ,...., parameters ?
And during downlink (iOS -> peripheral) ?
Cheers,
Alexis

philips77, 2015-05-30:
Hi,
As far as I remember all iPhones have the lowest interval connection 7.5ms, and send up to 4 packets each interval. The packet size is 20b. The speed depends also on the Packet Receipt Notification settings.
Regards,
Aleksander

alexisduque, 2015-05-30:
Hi Aleksander, 
I think iPhones have lowest interval connection 20 ms, no ? 
You can have a look at the [Apple iOS Bluetooth Design Guidelines](https://developer.apple.com/hardwaredrivers/BluetoothDesignGuidelines.pdf)
p. 22.
Am I wrong ?

philips77, 2015-05-30:
Hi,
Let me check this on Monday.
BR,
Aleksander

Kamran-Soomro, 2015-06-01:
The DFU feature inside nRF Toolbox app will let you know the number of bytes transferred and transfer time after uploading firmware.
Also you can set the Number of Packets to 25 from iPhone Setting  -> nRF Toolbox to increase the throughput.
On iPhone6, IOS 8.3, With Number of Packets set to 25, I have tried to transfer a firmware file of size 88344 bytes and it took 35 seconds so throughput is 20 kbps.

The normal connection interval set by IOS is 30ms, packets size is 20 bytes, and number of packets transferred per connection interval are approx. 6.

Please Let me know if you have other questions.

Regards,
Kamran

alexisduque, 2015-06-01:
Thank you Kamran :+1:

closed 2015-06-01 by alexisduque