nordicsemi/IOS-nRF-Toolbox #4: Split functionality into cocoapods deps

jacobrosenthal, 2015-11-11:
I needed DFU separate to bring into my app as a dependency so Im doing that here:
https://github.com/jacobrosenthal/nordic-ble-dfu

And here is a fork of this repo that uses it
https://github.com/jacobrosenthal/IOS-nRF-Toolbox/tree/pods

Willing to make this something PR -able if you're at all interested

huguesbr, 2016-02-09:
+1 interested.

@jacobrosenthal Could you document usage of your DFU Pod?

huguesbr, 2016-02-09:
I've also submitted a question on Nordic Developer Forum: https://devzone.nordicsemi.com/question/67419/ios-or-c-dfu-library/

jacobrosenthal, 2016-02-09:
I just copied their dfuviewcontroller into my app and edited to taste.

huguesbr, 2016-02-10:
Oh I see. I wanted to extract the actual DFU related functionalities (
dfuHelper, dfuOperations ) in its own library ( without any UI ).
I will ping back when I have something.

jacobrosenthal, 2016-02-10:
Seems totally doable.

huguesbr, 2016-02-10:
@jacobrosenthal for sure, but unfortunately not done (yet) in this App VS SDK. :)

philips77, 2016-02-10:
Hi,
We have published yesterday new version of nRF Toolbox. Should be out when Apple verifies that. The DFU part hasn't changed much, only some ui tweeks. I'll publish the code here when it's out.
Also we are working now on a new implementation of DFU in swift. We plan to released as framework in few weeks.

huguesbr, 2016-02-10:
Awesome. Thank you. Will the framework be more like a SDK/Library or still
an integrated App?

huguesbr, 2016-02-10:
@philips77 you said you've push new version of the Toolbox to Apple, any chance you could also update this Github or put the new app on a new repo? this repo haven't been updated since August 2015 :(

Thanks ^^

philips77, 2016-02-11:
Of course we will. As only Apple accepts the app.

philips77, 2016-02-11:
It will be released as a separate github project, only the framework. Like for Android.

huguesbr, 2016-02-11:
Top! Thanks

huguesbr, 2016-02-15:
@philips77 awesome, please keep us posted in this issue. Thx

philips77, 2016-02-15:
Hi, the source code for version 2.5 is now available. Have a great day, despite it's Monday ;)

huguesbr, 2016-02-15:
@philips77 great, you made my day a bit more like a Wednesday (not quiet a friday, yet) ^^ Will you also push the DFU related framework like the Android one (https://github.com/NordicSemiconductor/Android-DFU-Library)? Thx

philips77, 2016-02-15:
:)

We are working now on a new implementation of the DFU service, in swift. When done, we'll include it in nRF Toolbox and nRF MCP and publish it in a separate github repo available as framework. But I've just started it last week so it may take some time. The current implementation will stay as is.

huguesbr, 2016-02-15:
Got it. Thx.

philips77, 2016-03-03:
Hi, I've just published the source code for the DFU Library: https://github.com/NordicSemiconductor/IOS-DFU-Library 
You should find this framework much easier to use now. The nRF Toolbox 3.0 (also published a second ago on GitHub) shows how to use it. I'm really courious what do you think about the framework. Please, give any feedback in IOS-DFU-Library project's issue page.

closed 2016-03-03 by philips77

huguesbr, 2016-03-03:
@philips77 awesome :+1:  thx