nordicsemi/IOS-nRF-Toolbox #9: changes in version 2.5

foobar8675, 2016-02-16:
Apologies, this isn't an issue. Area all the changes with the new version visual? Or are there changes to anything that involves firmware updates?

I abstracted the fw update code from the pervious version so when looking at the changes for this version, there are a lot of diffs.

foobar8675, 2016-02-16:
Hello, I went through the diffs but wanted to know if you could confirm some things for me, and if / how they might affect bootloader/softdevice/app fw updates.

1  in the 2 links below, I see the new version is not getting the dfuVersion, but I don't think this should make a difference. Is there a reason it was removed?

https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/commit/6677cab0eb102ffd094026706e19b5fffe32d0ba#diff-884de5849b9ff76641ef73a97554afd4L124

https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/commit/6677cab0eb102ffd094026706e19b5fffe32d0ba#diff-884de5849b9ff76641ef73a97554afd4L143

2  this looks like a change I should make? has there been a change to the version number schemes?

https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/commit/6677cab0eb102ffd094026706e19b5fffe32d0ba#diff-884de5849b9ff76641ef73a97554afd4L466

3  this looks like a change i should make?

https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/commit/6677cab0eb102ffd094026706e19b5fffe32d0ba#diff-6cf52c8d210cb7cad58b37cd8537e15bL95

4   in DFUController, the changes I see are all related to upacking the manifest and logging. is that correct? 

https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/commit/6677cab0eb102ffd094026706e19b5fffe32d0ba#diff-0cbf51245064301a1a32deeb9bf76f6eL1

philips77, 2016-02-17:
Hi,
There were no changes in DFU. Only some polishing (better file selection) and adding better status reporting.
1. By the time if was reading the version there the version was already known. It reads it when it connects to the target, doesn't have to read again after pressing Upload button. I removed it from there mainly to get better 'status' messages for the user.
2. No, the DFU versions didn't changed. Now, when the DFU version char is not found, I returned -1 instead of 0, but it shouldn't matter. Neither of this versions should be ever returned from the DFU targ.
3. I guess yes. All for URLs should be nullify here.
4. Yes.

We are working now on a new DFULibrary, written in Swift, which will be available as a framework in a separate GitHub project. Planned features:
- API similar to Android's one: DFUServiceInitiator
- Sends all content of ZIP (SD+BL+APP) automatically (in 2 connections)
- Swift 2.1 or so
- Much more documentation and I think easier to understand.

ETA: ~2-3 weeks.

foobar8675, 2016-02-17:
Thank you @philips77 . Also, I'm very much looking forward to the swift DFULibrary!

closed 2016-02-17 by foobar8675