nordicsemi/IOS-nRF-Toolbox #12: readSFloatValue Bug for BGM

keithatci, 2016-03-24:
FYI in case anyone else runs into this problem...

Testing a True Metrix Air Blood Glucose Meter with this code does not work.  The meter is transferring per the profile standard, but the glucose concentration is not being calculated properly inside CharacteristicReader.m.  See correct implementation here...

https://github.com/signove/antidote/blob/master/src/util/bytelib.c

Also,  GlucoseReading.m updateFromBytes: is not checking the glucoseConcentrationUnit.  So multiplying by 1000 is not necessarily correct (not sure it ever makes sense??).  Should check whether unit is kg/L or mol/L before converting.

philips77, 2016-03-25:
Hi,
You are right, it looks like the parser ignores the unit at all.
It assumes the value is in mol/L. The value being displayed is in mmol/L, that's why it multiplies by 1000.
If the unit is kg/L it should convert it to mmol/L as well (as this value is shown on GUI). The MW of glucose is 180.1559 g/mol and density 1.54 kg/L (according to wiki), so that is 1540000 grams/m^3, so that is 1540000/180.1559 moles/m^3, so that is 1540/180.1559 moles/L, if I'm not mistaken. Assuming normal conditions... Please, correct me if it's wrong.
I'll fix it after Easter. Thanks for pointing it out.

philips77, 2016-03-25:
I'll also fix the sfloat parser.

keithatci, 2016-03-25:
OK that makes sense, I overlooked the mmol/l conversion. 

I'm not totally clear on the conversion from kg/l to mol/l.  Since you typically see these glucose measurements represented as mg/dl or mmol/l on the meter, I would just convert kg/l to mg/dl and convert mol/l to mmol/l.  

Thanks!

philips77, 2016-03-25:
Yes, but on the screen the unit is in a static label per table, not per row.. In theory each row may have a different unit. To convert from kg/L to mmol/L the MW and glucose density are required. But... Perhaps your are right, the unit should be per row.

philips77, 2016-03-31:
Hi, I just pushed changes on GitHub. The app on AppStore will be available in some time. Thanks again for the bug report.
Finally I decided to move the unit to each row and I'm not converting kg/l to mmol/l, but rather displaying value in mg/dl. I hope the conversion works fine now.

closed 2016-03-31 by philips77