nordicsemi/IOS-nRF-Toolbox #13: DFU file not support

yongmaowang, 2016-04-14:
With the default demo code about DFU, I got the message “Selected file is not supported.” after I selected hex file. Why?

mostafaberg, 2016-04-14:
Hi @yongmaowang !, I'll be working on this issue with you, so far we couldn't reproduce it though.

Was it one of the files that are bundled in the build or did you add a file yourself? if so can you give us the file name that causes the issue and perhaps more info on what you were trying to accomplish and what devices/versiosns you used ?

yongmaowang, 2016-04-15:
Hi @mostafaberg !, Thank you for responding. All the files in the build and include myself added, I got the same error "Selected file is not surpported!". My device is nRF51822, and the demo code run in iphone 6s plus with os 9.3.

mostafaberg, 2016-04-15:
Something sounds wrong, this does not reproduce on the same version here!
When you build the project from source without any modifications, do you see a list of bundled firmwares or do you have to add them manually ?
the files bundled should work out of the box, if you pull the project and build from scratch, do you see the list of firmwares ?

If it's possible to get access to your project I can check and help you a bit faster, if it's not possible we will just debug it together step by step 👍

mostafaberg, 2016-04-17:
Hi @yongmaowang!, Just checking in to see if you still have that problem or did you manage to resolve it ?

yongmaowang, 2016-04-18:
Hi @mostafaberg !, I will check in my project soon and give you the url.

yongmaowang, 2016-04-18:
Hi @mostafaberg !, My project is here: https://github.com/yongmaowang/IOS-nRF-Toolbox-master-wym

mostafaberg, 2016-04-18:
@yongmaowang awesome!, I'm looking into it now!

mostafaberg, 2016-04-18:
@yongmaowang I made a PR now, check it out and let me know if the problem is resolved!

yongmaowang, 2016-04-18:
@mostafaberg Thank you, If you have any progress, please let me know!

mostafaberg, 2016-04-18:
@yongmaowang No problems, will do that, ping back if the issue is resolved or not so I can close the issue if it's fixed:)

yongmaowang, 2016-04-18:
@mostafaberg I don't know if the problem is associated with the ios version. Now I still got that problem.

mostafaberg, 2016-04-18:
Strange, Cannot reproduce here, can you try another device with a different version ? what's the console output when this happens ? and do you have a screenshot of the UI state ?

yongmaowang, 2016-04-18:
I do not have another device now, mybe I will try later.

/**********Console Output Start*****************************/
2016-04-18 17:03:33.103 nRF Toolbox[1395:849252] getFilesFromDirectory
2016-04-18 17:03:33.104 nRF Toolbox[1395:849252] app resource path: /var/containers/Bundle/Application/7DB7B12D-9253-4895-A02E-CC0ECA77E76F/nRF Toolbox.app
2016-04-18 17:03:33.105 nRF Toolbox[1395:849252] firmware folder path: /var/containers/Bundle/Application/7DB7B12D-9253-4895-A02E-CC0ECA77E76F/nRF Toolbox.app/firmwares
2016-04-18 17:03:33.105 nRF Toolbox[1395:849252] getFilesFromDirectory
2016-04-18 17:03:33.105 nRF Toolbox[1395:849252] app resource path: /var/containers/Bundle/Application/7DB7B12D-9253-4895-A02E-CC0ECA77E76F/nRF Toolbox.app
2016-04-18 17:03:33.105 nRF Toolbox[1395:849252] firmware folder path: /var/containers/Bundle/Application/7DB7B12D-9253-4895-A02E-CC0ECA77E76F/nRF Toolbox.app/firmwares
/**********Console Output End*****************************/

[image]

mostafaberg, 2016-04-18:
Interesting bug there, one last request, can I see a screenshot of what it looks like when you're selecting from the firmware list ? i'll try to look deeper into that but it seems like it's something with that device

yongmaowang, 2016-04-18:
[image]
[image]

philips77, 2016-04-18:
Hi, are you using the nRF Toolbox available on App Store? Version 3.0? The previous one, 2.5, or you compiled the newest source from GitHub (3.0.1)? In the 3.0 I think there is a but and those files are not visible on the App Files tab at all. Are you able to debug and see why the DFUFirmware doesn't get created?

philips77, 2016-04-18:
In the DFUFirmware class, in the DFU Library, there is a method init?(urlToBinOrHexFile:urlToDatFile:type). The message you get would be triggered it this method returns null, and this may happen only if the file name does not end with 'hex' or 'bin'. Could you place a breakpoint there and see when it returns?

yongmaowang, 2016-04-18:
I will try later, and thank you very much!

mostafaberg, 2016-06-07:
Hi @yongmaowang , any updates on this issue ?

closed 2016-09-05 by mostafaberg