nordicsemi/IOS-nRF-Toolbox #14: App crashes at splash screen

samueleforconi, 2016-04-19:
Hi,
I upgraded the nRF-Toolbox app to version 3.0 on 2 devices: iPad Mini (iOS 9.3.1), iPhone 4S (iOS 8.4).
On both devices the app crashes on the splash screen, no messages are displayed, I'm not able to use the app anymore, please solve the bug or tell us how to solve.
We are beacon developers/manufacturers, we use the app to test the FW upgrade procedure on our beacons.

mostafaberg, 2016-04-19:
Hi @samueleforconi sorry to hear that you're having an issue with our app, we need a bit more information so we can help you further, can you open up XCode's console and enable breakpoints and show us where the app crashes ?

mostafaberg, 2016-04-19:
But first things first, can you try uninstalling the app, rebooting your device then doing a fresh install and see if that helps ? one of our devices here crashed while testing then it worked when we did a fresh reinstall/reboot. let me know if that helps, we couldn't get any crash logs from the device though so it might be an iOS bug, keep us updated!

samueleforconi, 2016-04-19:
Hi,
I tried the following: 
1. remove the app and reinstall (without rebooting the iPad), still crash. 
2. remove the app, reboot the iPad, reinstall the app, still crash.
3. compile and run the app on the iPhone, crash with log:
`dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /private/var/mobile/Containers/Bundle/Application/420F3E48-9B9F-474A-BA1C-306169DF67A5/nRF Toolbox.app/Frameworks/DFULibrary.framework/DFULibrary
  Reason: Incompatible library version: DFULibrary requires version 1.0.0 or later, but libswiftCore.dylib provides version 0.0.0`

mostafaberg, 2016-04-19:
Seems like a swift versioning issue, what version of XCode are you using ?

samueleforconi, 2016-04-19:
I'm using XCode 7.2, now downloading 7.3 and try to run the app with the latest build of XCode.

mostafaberg, 2016-04-19:
Yeah that sounds good, if it still doesn't work can you try build on a device with iOS 9.0 or above ? not 100% sure about iOS 8 compatibility now, sorry about that!

mostafaberg, 2016-04-19:
@samueleforconi Any updates ? I assume it worked with Xcode 9.3 since that bundles the latest version of swift, lest us know if it worked or not 👍

samueleforconi, 2016-04-19:
Installing XCode 7.3 seems to be a long task... I'll keep you informed as soon as I'm able to repeat the test with XCode 7.3.

samueleforconi, 2016-04-19:
@mostafaberg 
Compiling the app with XCode 7.3 works on both devices (iPad-9.3.1, iPhone 8.4).
I still haven't an explanation on why the application updated/downloaded from the app store didn't work.

mostafaberg, 2016-04-20:
We will review this issue and address in the next release !, thanks for your help 👍

closed 2016-04-20 by mostafaberg