nordicsemi/IOS-nRF-Toolbox #17: invalid state error

foobar8675, 2016-04-21:
when doing a DFU update with a single zip that contains an application, softdevice, and bootloader, I am sometimes getting an Invalid State Error, even if the device is on the charger.

[image]

this is running the latest code out of master. i'm not able to consistently reproduce. any help with understanding when this error appears would be helpful. thanks.

philips77, 2016-04-21:
Check out this issue: https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/8
Maybe adding 1.5 sec fixes the issue.

foobar8675, 2016-04-21:
Thanks @philips77 . to double check, the fix might be apply a 1.5 s delay before calling my equvalient of https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/master/nRF%20Toolbox/DFU/DFUViewController/DFUViewController.m#L132 ?

i'll give that a shot. but what is pretty interesting is i did not have this issue occur with the current version in the app store. is there  a delta between the two?

philips77, 2016-04-22:
The complete source is available here. The current version from master will be available as 3.0.1 soon, but you'll find a commit with version 3.0. But I see no reason why would such a difference in behavior exists.

foobar8675, 2016-04-28:
Hi @philips77 I'm not 100%, since my failures are intermittent, but I'm trying things with the 1.5 second delay and I 'think' it seems better (as per your suggestion) . Do you have any thoughts as why that would make a difference?

philips77, 2016-04-28:
It may be related to the flash operations timings. It depends when do you get this error, just after sending Start DFU command and sizes? Or later. You may ask this question on the DevZone, at nordicsemi.com. Our support team will be much more qualified to speak about the bootloader.

mostafaberg, 2016-09-06:
@foobar8675 Sorry to refresh the old issue, did you get this working ? we have just released v4.0.4 and would like you to try it and see if your issue still reproduces, I'll give it a couple of days before closing this issue since it's been "resolved" for a few months now  👍 , just feel free to reopen if you still encounter issues later

liuyan3176, 2016-09-06:
Sometimes also meet this issue...
And I found the reason is my phone try to re-connect the device nor to update it, so the device is in invalid state.  
But I also have a little problem,when I update the device successful(StateCompleted),I have to re-scan the retrieveConnectedArray,wait the device name appear and can be connected.
Does anything matter in my update  process?Is there anything method to auto-connect the device after the updating is successful? 
Thanks a lot ~

philips77, 2016-09-06:
Fyi. If the previous DFU operation failed during uploading and you try to repeat it, the device will be in a state that allows resuming only. Sending Start DFU command will respond with Invalid State. The next retry will succeed.
We don't have resuming implemented in legacy DFU in ios nor Android. However, since 0.6.3(?) Android DFU library will try to reconnect itself if Invalid State error received.

mostafaberg, 2016-09-06:
@liuyan3176 Your issue is different it seems, first of all the library is only responsible for flashing, anything happening after that is up to you as a developer.

regarding the invalid state is that if you disconnect you will have to restart the whole process since we don't support resuming in the Legacy DFU mode.

foobar8675, 2016-09-08:
@mostafaberg we have a delayed timer which seems to fix things but can you share what is different in 4.0.4 that would make this behave better?

qszdjj, 2016-09-29:
Version 4.0.4(12). IOS 10 The above all this problem.

mostafaberg, 2016-10-04:
@foobar8675 Can you take a look at [today's release](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/releases/tag/4.1.0) and see if there's anything different than what you're doing ?

foobar8675, 2016-10-04:
Hi, I'm not working on this project anymore, but am cc'ing a couple people
who are.

mostafaberg, 2016-10-05:
that's great, keep us posted, if we don't hear back in a couple of days we'll assume issue is resolved, you can ask the new person responsible to re-open a new issue if the bug is still there

mostafaberg, 2017-02-06:
Issue resolved ? :)

closed 2017-03-13 by mostafaberg