nordicsemi/IOS-nRF-Toolbox #20: Scanner Delegate is not working.

imanupgupta, 2016-06-06:
Hi,
when i m using only **Proximity** . when I use
 `- (void) centralManager:(CBCentralManager*) manager didPeripheralSelected:(CBPeripheral*) peripheral;`
this method in **ProximityViewController** it not work. because when this method is calling from **ScannerViewController** it not redirect to me at **ProximityViewController** so i can't understand what should i do now.

mostafaberg, 2016-06-06:
Hi @imanupgupta, is it not working in our example app out of the box?
Or is it not working in your own implementation ?

imanupgupta, 2016-06-06:
Hi @mostafaberg  its not working in my project when i am implement this. its proper work in your example.

mostafaberg, 2016-06-06:
aha I see, so this is not an issue with this code, please post your question in the Nordic developer zone so we can further help you with your personal code issues, i would suggest looking into your delegates as you might be missing something.

closed 2016-06-06 by mostafaberg