nordicsemi/IOS-nRF-Toolbox #21: nRF Toolbox DFU Operation failed

LtkDev, 2016-08-18:
With the default demo code or download nRF Toolbox App from AppStore, it's fail.
Does anyone know what the problem might be? Thanks
devices  ：nRF52 demo board PCA10040 V 1.1.0
mobile    ：iPhone 5s, iOS 8.1.3  /  iPhone 6Plus, iOS 9.3.3

### step1:Enter bootloader DFU mode.
[image]

### step2:Select file "hrs_dfu_s130_2_0_0_7a_sdk_11_0_0_2a.zip", and upload.
[image]

### step3:Get the error message "DFU Operation failed", and log as following.
[image]
objc[860]: Class BluetoothManager is implemented in both /System/Library/PrivateFrameworks/BluetoothManager.framework/BluetoothManager and /private/var/mobile/Containers/Bundle/Application/81EEFFC7-2F0D-4212-8BA6-365D8407B50E/nRF Toolbox.app/nRF Toolbox. One of the two will be used. Which one is undefined.
2016-08-18 16:09:20.112 nRF Toolbox[860:79822] getFilesFromDirectory
2016-08-18 16:09:20.115 nRF Toolbox[860:79822] app resource path: /private/var/mobile/Containers/Bundle/Application/81EEFFC7-2F0D-4212-8BA6-365D8407B50E/nRF Toolbox.app
2016-08-18 16:09:20.116 nRF Toolbox[860:79822] firmware folder path: /private/var/mobile/Containers/Bundle/Application/81EEFFC7-2F0D-4212-8BA6-365D8407B50E/nRF Toolbox.app/firmwares
2016-08-18 16:09:20.116 nRF Toolbox[860:79822] getFilesFromDirectory
2016-08-18 16:09:20.117 nRF Toolbox[860:79822] app resource path: /private/var/mobile/Containers/Bundle/Application/81EEFFC7-2F0D-4212-8BA6-365D8407B50E/nRF Toolbox.app
2016-08-18 16:09:20.117 nRF Toolbox[860:79822] firmware folder path: /private/var/mobile/Containers/Bundle/Application/81EEFFC7-2F0D-4212-8BA6-365D8407B50E/nRF Toolbox.app/firmwares
2016-08-18 16:09:23.431 nRF Toolbox[860:79822] 
WARNING: The class 'Manifest' is not key value coding-compliant for the key 'dfu_version'
 There is no support for optional type, array of optionals or enum properties.
As a workaround you can implement the function 'setValue forUndefinedKey' for this. See the unit tests for more information
2016-08-18 16:09:29.934 nRF Toolbox[860:79822] 1: Connecting to DfuTarg...
2016-08-18 16:09:29.935 nRF Toolbox[860:79822] 0: centralManager.connectPeripheral(peripheral, options:nil)
2016-08-18 16:09:30.107 nRF Toolbox[860:79893] 0: [Callback] Central Manager did connect peripheral
2016-08-18 16:09:30.108 nRF Toolbox[860:79893] 5: Connected to DfuTarg
2016-08-18 16:09:30.109 nRF Toolbox[860:79893] 1: Discovering services...
2016-08-18 16:09:30.109 nRF Toolbox[860:79893] 0: periphera.discoverServices(nil)
2016-08-18 16:09:30.531 nRF Toolbox[860:79903] 5: Services discovered
2016-08-18 16:09:30.532 nRF Toolbox[860:79903] 1: DFU Service found
2016-08-18 16:09:30.534 nRF Toolbox[860:79903] 1: Discovering characteristics in DFU Service...
2016-08-18 16:09:30.535 nRF Toolbox[860:79903] 0: peripheral.discoverCharacteristics(nil, forService:DFUService)
2016-08-18 16:09:30.804 nRF Toolbox[860:79893] 5: DFU characteristics discovered
2016-08-18 16:09:30.807 nRF Toolbox[860:79893] 1: Reading DFU Version number...
2016-08-18 16:09:30.808 nRF Toolbox[860:79893] 0: peripheral.readValueForCharacteristic(00001534-1212-EFDE-1523-785FEABCD123)
2016-08-18 16:09:30.863 nRF Toolbox[860:79903] 5: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):0800
2016-08-18 16:09:30.864 nRF Toolbox[860:79903] 10: Version number read: 0.8
2016-08-18 16:09:30.865 nRF Toolbox[860:79903] 1: Enabling notifiactions for 00001531-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:30.865 nRF Toolbox[860:79903] 0: peripheral.setNotifyValue(true, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123)
2016-08-18 16:09:30.981 nRF Toolbox[860:79893] 1: Notifications enabled for 00001534-1212-EFDE-1523-785FEABCD123
2016-08-18 16:09:30.982 nRF Toolbox[860:79893] 10: DFU Control Point notifications enabled
2016-08-18 16:09:30.983 nRF Toolbox[860:79893] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:30.986 nRF Toolbox[860:79893] 0: peripheral.writeValue(0x0103, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-08-18 16:09:30.988 nRF Toolbox[860:79893] 1: Writing image sizes (104164b, 12996b, 0b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:30.990 nRF Toolbox[860:79893] 0: peripheral.writeValue(0xe4960100c432000000000000, forCharacteristic: 00001534-1212-EFDE-1523-785FEABCD123, type: WithoutResponse)
2016-08-18 16:09:31.041 nRF Toolbox[860:79903] 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2016-08-18 16:09:31.042 nRF Toolbox[860:79903] 10: Start DFU (Op Code = 1, Upload Mode = 3) request sent
2016-08-18 16:09:31.582 nRF Toolbox[860:79893] 5: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100101
2016-08-18 16:09:31.583 nRF Toolbox[860:79893] 10: Response (Op Code = 1, Status = 1) received
2016-08-18 16:09:31.584 nRF Toolbox[860:79893] 10: Writing Initialize DFU Parameters...
2016-08-18 16:09:31.584 nRF Toolbox[860:79893] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:31.585 nRF Toolbox[860:79893] 0: peripheral.writeValue(0x0200, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-08-18 16:09:31.586 nRF Toolbox[860:79893] 1: Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:31.588 nRF Toolbox[860:79893] 0: peripheral.writeValue(0xffffffffffffffff0100feff7630, forCharacteristic: 00001534-1212-EFDE-1523-785FEABCD123, type: WithoutResponse)
2016-08-18 16:09:31.590 nRF Toolbox[860:79893] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:31.591 nRF Toolbox[860:79893] 0: peripheral.writeValue(0x0201, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-08-18 16:09:31.640 nRF Toolbox[860:79903] 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2016-08-18 16:09:31.700 nRF Toolbox[860:79893] 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2016-08-18 16:09:31.701 nRF Toolbox[860:79903] 5: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100201
2016-08-18 16:09:31.701 nRF Toolbox[860:79903] 10: Response (Op Code = 2, Status = 1) received
2016-08-18 16:09:31.701 nRF Toolbox[860:79903] 10: Initialize DFU Parameters completed
2016-08-18 16:09:31.702 nRF Toolbox[860:79903] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:31.703 nRF Toolbox[860:79903] 0: peripheral.writeValue(0x080c00, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-08-18 16:09:31.760 nRF Toolbox[860:79893] 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2016-08-18 16:09:31.761 nRF Toolbox[860:79893] 10: Packet Receipt Notif Req (Op Code = 8, Value = 12) request sent
2016-08-18 16:09:31.761 nRF Toolbox[860:79893] 1: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
2016-08-18 16:09:31.762 nRF Toolbox[860:79893] 0: peripheral.writeValue(0x03, forCharacteristic: 00001531-1212-EFDE-1523-785FEABCD123, type: WithResponse)
2016-08-18 16:09:31.821 nRF Toolbox[860:79903] 5: Data written to 00001531-1212-EFDE-1523-785FEABCD123
2016-08-18 16:09:31.821 nRF Toolbox[860:79903] 10: Receive Firmware Image (Op Code = 3) request sent
2016-08-18 16:09:31.822 nRF Toolbox[860:79903] 10: Uploading firmware...
2016-08-18 16:09:31.822 nRF Toolbox[860:79903] 1: Sending firmware DFU Packet characteristic...
2016-08-18 16:09:31.881 nRF Toolbox[860:79893] 5: Notification received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x):100306
2016-08-18 16:09:31.882 nRF Toolbox[860:79893] 10: Response (Op Code = 3, Status = 6) received
2016-08-18 16:09:31.883 nRF Toolbox[860:79893] 20: Error 6: Operation failed
2016-08-18 16:09:31.883 nRF Toolbox[860:79893] 1: Disconnecting...
2016-08-18 16:09:31.884 nRF Toolbox[860:79893] 0: centralManager.cancelPeripheralConnection(peripheral)
2016-08-18 16:09:31.887 nRF Toolbox[860:79903] 0: [Callback] Central Manager did disconnect peripheral without error
2016-08-18 16:09:31.887 nRF Toolbox[860:79903] 5: Disconnected
2016-08-18 16:09:31.888 nRF Toolbox[860:79822] Error 6: Operation failed

mostafaberg, 2016-08-18:
Interesting find, what kind of device are you trying to flash ?

philips77, 2016-08-18:
1. You are flashing s130 onto nRF52 device which is not compatible. For nRF52 the S132 should be used. But this should not matter, I guess, as the DFU bootloader does not know it yet. It would just brick* your device if successful.
2. Could you try with lower value of Packet Receipt Notification? You can set it in nRF Toolbox settings in Settings application. Set value to 1 first, then try higher values.
* - semi-brick, as you would be able to flash correct version using USB

LtkDev, 2016-08-19:
Hi philips77, thank a lot!
I had try S132 or my zip file, it was still failed. Through breakpoints, it's failed in DFUService.swift 
"func sendFirmware()", and got ErrorCallback as following.
> (lldb) print report
(iOSDFULibrary.ErrorCallback) $R0 = 0x000000010041e410 iOSDFULibrary`partial apply forwarder for iOSDFULibrary.DFUPeripheral.(sendFirmware (iOSDFULibrary.DFUFirmware, withPacketReceiptNotificationNumber : Swift.UInt16, andReportProgressTo : Swift.Optional<iOSDFULibrary.DFUProgressDelegate>) -> ()).(closure #1).(closure #2) at DFUPeripheral.swift

> 
[image]

mostafaberg, 2016-08-21:
@LtkDev is that crash using in that screenshot using our own bundled firmwares or yours ?

Please do the following so we can confirm where the problem is

1-Remove any bonded peripherals in the bluetooth settings.
2-Uninstall your version of nRF Toolbox.
3-Install nRF Toolbox from the app store.
4-Restart phone.
5-Launch nRF Toolbox.
6-Put your peripheral in DFU Mode
7-Select the bundled firmware `hrs_dfu_s132xxx.zip`

If that still doesn't work then something is probably wrong with your board, please try to flash it with nRF GO on your desktop with a compatible firmware and try again, let us know how things went

LtkDev, 2016-08-25:
@mostafaberg Thank you so much!  DFU is workable!
After reinstall nRF Toolbox and use another board with SDK v11 & s132 v2.0.0, it worked now.

mostafaberg, 2016-08-26:
Glad to hear it worked out!, let us know if there's anything else we can help with, for now I'll close this issue 👍

closed 2016-08-26 by mostafaberg