nordicsemi/IOS-nRF-Toolbox #23: Crashing during DFU

Redguy1986, 2016-09-14:
Version 4.0.4
I choose DFU mode, then Select any zip-file, then push button "Select Device". The app crashes down.
It does not matter if I don't choose any file or choose hex-file
When I choose UART mode, I have the same behavior after "Connect" button pressed.

develooop, 2016-09-16:
Version 4.0.4(12). iPhone SE(IOS 9.3.2) and 6S(IOS 10.0.1)
I sent my zip/hex file from computer to cell phone by email as an attachment. Then I opened the zip/hex file in my mailbox and pressed the botton ‘copy to nRF Toolbox’. When the page turned to app 'nRF Toolbox’    , the app 'nRF Toolbox’ crashed down immediately.

philips77, 2016-09-21:
Hi, we will look into it. Thanks for the report.

mostafaberg, 2016-09-23:
We are aware of this issue, the easiest workaround before we release the fix is to copy the files to the nRF Toolbox app using iTunes connect.

if this is not an option then please give us a few days and the fix will be out this coming working week, sorry for the delays!

mostafaberg, 2016-09-23:
@develooop please open up a new issue with your problem, this is not related to the open issue. i will take it from there.

mostafaberg, 2016-10-04:
Resloved

Redguy1986, 2016-10-07:
I've checked new version (4.1.0) from AppStore.
This issue didn't resolve!
The app still crashes when push button "Select Device" in DFU mode.

mostafaberg, 2016-10-07:
It's ok i've reopened it for you, The issue does not reproduce at all for us, can you try it on a different phone  ?

mostafaberg, 2016-10-07:
Great, will take it on from there in the [new issue](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/issues/26)

closed 2016-10-07 by mostafaberg