nordicsemi/IOS-nRF-Toolbox #26: Crashing during DFU

Redguy1986, 2016-10-07:
Version 4.1.0. iPhone 5c, IOS 9.3.4
I choose DFU mode, then push button "Select Device". The app crashes down.
I choose UART mode, I have the same behavior after "Connect" button pressed.
I tried it in app from official AppStore

Redguy1986, 2016-10-07:
When I tried the app on iPhone 5s with iOS 10 it works properly.

mostafaberg, 2016-10-07:
Ok let's see, you're saying it crashes only on iOS10 ?

Redguy1986, 2016-10-07:
No.
It crashes on iOS 9.3.4.
I've tried it on iOS 10, and the app works well.
So, it depends on iOS version

mostafaberg, 2016-10-07:
Ok so it works ok here on iOS 9, testing on 10 to verify, but if it works here on iOS 9 ..

Can you help out by attaching your device crash logs ?

Redguy1986, 2016-10-07:
How can I help you?
I tried to get logs in iTools program.
Look at these files:
[itools_log.txt](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/files/515606/itools_log.txt)
[nRF Toolbox-2016-10-07-121411.ips.txt](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/files/515607/nRF.Toolbox-2016-10-07-121411.ips.txt)
[nRF Toolbox-2016-10-07-121438.ips.txt](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/files/515605/nRF.Toolbox-2016-10-07-121438.ips.txt)
Are they useful for you?

mostafaberg, 2016-10-07:
They are not symbolicated so it's a bit hard to know what happened exactly.
Can you plug it into XCode, and export the last crash report from the device manager there and send me that file ? it's ok if that's not symbolicated too as long as you are using the exact last version form the app store

Redguy1986, 2016-10-07:
I'm sorry, I can't use Xcode. I'm an application user, not an iOS developer. Besides, I use Windows7 on my PC and I have no possibility to use apple features or programs.

mostafaberg, 2016-10-07:
Ah ok, thought you had a development environment, ok i'll try to reproduce here and see what I can do, it doesn't reproduce that's the problem, but will experiment and try to see if I can resym your crashes so I can understand what's going on, will keep you updated.
Thanks a lot for your help!

paweljankowski, 2016-10-26:
This crash occurs on iOS 10.1 (on iOS 10 works perfectly). I was able to download sources and launch nRFTToolbox and the issue is with class `NORScannerViewController` line 168:
```
Could not cast value of type 'NSKVONotifying_CBPeripheral' (0x174300bd0) to 'nRF_Toolbox.NORScannedPeripheral' (0x100163e28).
2016-10-26 08:34:16.326754 nRF Toolbox[2206:577147] Could not cast value of type 'NSKVONotifying_CBPeripheral' (0x174300bd0) to 'nRF_Toolbox.NORScannedPeripheral' (0x100163e28).
```

Redguy1986, 2016-10-27:
When I updated iOS to 10 version this issue was still there.
Right now I've updated NRFToolbox to 4.1.1 version from appstore and the issue disappeared.
Thank you.

paweljankowski, 2016-10-27:
On iOS10.1 crash still exists.

closed 2016-11-07 by philips77

paweljankowski, 2016-11-21:
Can you release this fix to the AppStore?

philips77, 2016-11-21:
We will publish it tomorrow or on Wednesday, sorry for the delay, but I'm also working on new version of DFU library and need to finish it before.

PR #30 (2016-11-07): Converting NORScannerViewController to use a NORScannedPeripheral
Fixes #26

To repro issue before the fix:
- Go to UART, tap Connect, choose a device to connect to
- Now go back to the main screen with icons
- Choose UART again
- Tap Connect
- It crashes 100% because `getConnectedPeripherals` returns a non-empty array of `CBPeripheral` values instead of `NORScannedPeripheral` values

This diff takes the value of `getConnectedPeripherals` and converts it to an array of `NORScannedPeripheral` items
commit 3ae84d3a (2016-11-07): Converting NORScannerViewController to use a NORScannedPeripheral array, fixes #26