nordicsemi/IOS-nRF-Toolbox #27: nRFToolsBox - HTS issue

atlaschen1113, 2016-10-07:
Hi
In ble_app_hts sample code, the default setting

#define MIN_CELCIUS_DEGREES             3688

#define MAX_CELCIUS_DEGRESS             3972

#define CELCIUS_DEGREES_INCREMENT       36

When nRFtoolBox - HTS connect with Device, it will shows 3688+36 = 3724. It means 37.24 degrees

Android will show 37.24 degress but iOS will show 176.00 degress.

When press button 1 on nRF51 DK. It will increse 36.

Android will show 37.60 degress but iOS will show 212.00 degrees.

Customer wants to demo their HTS device to their end customer in these days.

When will Nordic release new version to fix this issue?

Thanks.

mostafaberg, 2016-10-07:
Thanks for your report!, I can confirm this is a bug on the iOS app and will be rectified this coming week!, and problem will also be resolved on the next Release to the app store.

Sorry for the inconvenience

mostafaberg, 2016-10-10:
Issue is now resolved from version `4.1.1`
We are in the process of submitting a new release to the app store, in the mean time you may compile and run this release instead, sorry for the inconvenience!

closed 2016-10-10 by mostafaberg