nordicsemi/IOS-nRF-Toolbox #28: BLE-UART String Send

lauleehong, 2016-10-08:
Hi,

Tested using iOS 10.1

Connecting to my device on UART-BLE, and then sending a string 'ACK' or any arbitrary characters would send up with a random hexstring sent. Sending the same character string 'ACK' twice shows on the log 2 different hexstrings are sent: for example "0x000abc" and "0x000abd". 

I've tested my hardware and firmware against the Android version of the nRF Toolbox, and is proven to be working normally.

mostafaberg, 2016-10-11:
Hi !, sorry to hear that, can you send us some concrete values we can test against? and show what should be displayed and what was displayed instead, will take a look into it when i have something to test against!, thanks!

lauleehong, 2016-10-12:
Hi, thanks for looking into this. 
Below is a screenshot from my iPhone: 

[image]
[image]

philips77, 2016-10-13:
@mostafaberg could you look into the log here, it should not print
'Optional', and should print text as string on Application level and as
bytes on debug.

mostafaberg, 2016-10-13:
@lauleehong Thanks for the detailed screenshot, will look int to that soon !
@philips77 Noted, will get that done too!

PR #31 (2016-11-07): UART string sending and receiving
Fixes #28 

Makes sending UART strings compatible with the new Swift 3 `UnsafeMutableRawPointer` and fixes problem where the pointer value is sent instead of the string value for strings that are shorter than 20 characters.

Also fixes occasional scroll to bottom crash in `NORLogViewController.swift`
commit 79f703a4 (2016-11-07): Fixes #28: UART string sending and receiving
closed 2016-11-07 by philips77