nordicsemi/IOS-nRF-Toolbox #32: iOS DFU not working since update

sanderdb01, 2016-11-03:
Hi, it seems that after the ios 10 update and the swift code conversion, my DFU code is not longer working correctly. In code, I am able to get the module into the DFU mode (DFTTarg), after that I am able to connect to the module again under the new service UUID for the DFU. However, after this step I try and call the DFUServiceInitiator.start() method, and it seems to work, but the only update I get (didStateChangedTo) is a state change to "connecting" and that is it, it stays there. I am able to do the DFU with an android phone using Nordic NRF Connect just fine, so the module is ok. Has the implementation of the code changed? Am I missing something? Is there example code I can refer to? Thanks a bunch, and if there is anything else that I could provide that might help, please let me know. Thanks.

mostafaberg, 2017-02-06:
Hi @sanderdb01, did you resolve this issue yet ? I assume you got help in the `iOSDFULibrary` issues right?

sanderdb01, 2017-02-06:
Sorry for the radio silence, things got crazy on my end. Yes everything
worked out great, and thank you so much for all the help. We pushed the
update to the users, so they should all be happy now too. If there are any
other issues, I will let you know.

Thanks again

mostafaberg, 2017-02-07:
Awesome !, please let us know if there's anything we can help with :)

closed 2017-02-07 by mostafaberg