nordicsemi/IOS-nRF-Toolbox #33: DFU not working

andrewbollen, 2016-11-06:
On pressing upload, app crashes and the configurable settings (number of packets and External MCP DFU) are reset to 12 and OFF respectively. We require number of packets to be 1, and External MCP DFU to be ON. We did not have this issue in v3.0.1.

mostafaberg, 2017-02-06:
Hi @andrewbollen , can you try updating to the latest DFU Library version in the app and rebuild ?
modify the Podfile so this line:
`pod 'iOSDFULibrary', :git => 'https://github.com/nordicsemiconductor/ios-pods-dfu-library', :branch => 'Version3'`

becomes:

`pod 'iOSDFULibrary', '~> 3.0'`

then run `pod update`

if the issue still persists please let us know

andrewbollen, 2017-02-06:
Thankyou @mostafaberg, the latest update to nRF-Toolbox does not appear to have the same problem, and we have been able to correctly upload our firmware. Thanks for the response.

closed 2017-02-06 by andrewbollen