nordicsemi/IOS-nRF-Toolbox #35: Cryptic error message

huguesbr, 2017-04-27:
We wasted a valuable time trying to understand the reason of a issue we were having with OTA.

We were using NRF51 SDK 11 which support only legacy OTA.
Nordic documentation didn't specify anything about a specific version of `nrfjprog`, but we learn (via amazing Nordic Support) that we needed to use `nrfutil` <= `0.5.2` (https://github.com/NordicSemiconductor/pc-nrfutil/tree/v0.5.2) in order to generate a legacy compatible package. 

I'm not sure if the error code return by the app match any other similar issue but we get "Operation failed. Ensure the firmware targets that device type and version." (i.e..: 
https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/0aae864a1ce0d34ac7f3cd589625d353f552d9be/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift#L293 ) which is pretty cryptic.

It would have been awesome to have something like: "This package version doesn't seem to be supported by your current boot loader SDK version. Please consider upgrading your SDK or use the correct firmware packaging format, see docs xxx for more help"

Thank you for your help and support, we know it's hard to maintain past documentation but I think legacy OTA (even if unsecure) is still pretty widely used for device which don't need this security.

Thanks and hope it helps others :)

philips77, 2017-04-27:
Hi, it's not that simple. None of the dfu libs (Android or iOS) is looking into the init packet. Its length also depends on the impl and users are free to add there their own validation data, so it can't also check the length, unless there is no DFU Version characteristic (very old sdk), where init packet could have contain only 16bit CRC. So the libs don't look inside. If the init packet was created for Secure DFU, but the legacy one is used, the init packet is sent and the legacy bootloader decodes it on its way, so expects some weird validation values. And it returns an error that the device type/revision or sd id does not match. That's where the error message comes from.
On the nrfutil GitHub page it is written that 2.0+ should be used only for Secure DFU, but nevertheless sorry for wasting your time...
Good luck with the project.

huguesbr, 2017-04-27:
Of course, sorry if I implied that there was an easy solution. I bet
dealing with a lot of version can be a hassle.

I was more suggesting as it's not really a end user app but more a
developer solution it could have express more lengthily error message
pointing potential solution.

Or maybe the app could ask what DFU/OTA version the device is expected to
run to do some validation, once again maybe not. DFU can be pretty obscure
so anything can go wrong and it's very to even start having an idea of what
is wrong.

Thanks and happy that it work now and hope this issue even if close will
let a mark for others :).

closed 2017-12-14 by mostafaberg

venkatesh-av, 2024-12-20:
> We wasted a valuable time trying to understand the reason of a issue we were having with OTA.
>
> We were using NRF51 SDK 11 which support only legacy OTA. Nordic documentation didn't specify anything [...]

thanks for the info. 

iam also using the legacy DFU. 

but now iam unable to install the nrfutil 5.2 or older verstions. is there any other way to geneare the zip file 
which supports the legacy DFU. Thanks in advace..!

philips77, 2024-12-20:
Hello,
You should still be able to install it using from https://pypi.org/project/nrfutil/0.5.3/ using `pip install nrfutil==0.5.3`.
Mind, that you may need Python 2.7 or the name may collide with the new version or new app (https://www.nordicsemi.com/Products/Development-tools/nRF-Util).

You may also generate it manually using https://github.com/NordicSemiconductor/Android-nRF-Connect/tree/main/init%20packet%20handling (old, but your legacy DFU is also old).