nordicsemi/IOS-nRF-Toolbox #41: IOS TOOLBOX DFU upload failure

ArthurHCL, 2017-12-13:
toolbox which version is newly on IOS,DFU can not work,I have found the reason...at upload start,toolbox does NOT write CCCD of DFU control point,so device think it is a failure,then RESET.
could you solve the problem of toolbox?

mostafaberg, 2017-12-14:
Hi Arthur, can you send us example firmware that reproduces this issue ? and perhaps the exact logs so we can know what the exact issue is ?

Paluche, 2017-12-21:
Hi @mostafaberg and @ArthurHCL I think I have the same problem, that could be seen also with the DFU on the nRF connect iOS application. The two applications can't activate the notifications on the Control Point characteristics. nRF Connect provides more informations on this matter:
[image]

I haven't been able to test with earlier versions of iOS. But the code I run on the embedded part works fine when running the DFU on the nRF Android applications.

philips77, 2017-12-21:
Hi, I see there 2 issue: as far as I know, the current DFU doesn't support higher MTUs, but your support 185+3, is it a custom modified one?
Looks like your control point characteristic doesn't have the required Client Characteristic Configuration descriptor, or there is an issue with caching. Did you try if it works on Android in nRF Connect?

ArthurHCL, 2017-12-22:
@mostafaberg I have solved this problem: I mistakenly used the HEX file directly for the DFU without adding the INIT package ... Now that my DFU is working perfectly, thanks!

Paluche, 2017-12-22:
@philips77 Yes it is a custom DFU and yes it works fine with the nRF Connect Android application. 
The custom DFU is basically a copy paste from the SDK sources.

mostafaberg, 2018-01-12:
@ArthurHCL Glad to hear it works for you!
@Paluche Can you open up a new issue so we can support you there instead ? thanks, closing this for now as it's resolved

closed 2018-01-12 by mostafaberg