nordicsemi/IOS-nRF-Toolbox #44: BGM Module

JunJun1994, 2018-01-04:
Can I download the engineering connect my yuwell glucose meter can obtain after the first and last data, but can't get all the data, I saw the break point, method of adding data doesn't seem to go, can you help me to solve it for me? Thank you very much!

philips77, 2018-01-04:
Did you try using nRF Connect to get your data? Could you paste here the log, and what would you expect to see?

JunJun1994, 2018-01-04:
Get the first data：
New glucose reading
OpCode: 6, Operator: 1
Get all the data：
OpCode: 6, Operator: 1

mostafaberg, 2018-01-12:
Can you show some example code and expected results and what you get instead ?

JunJun1994, 2018-01-12:
Thank you very much for your reply, I didn't change any of your source code, run your project directly and connect my glucose meter, and then obtain the glucose meter data, but sometimes can obtain all the data, sometimes can't, for the first and the last one is ok, look at the development of the document, to obtain all data is indeed "0 x01", the last one is "0 x06". The problem is, getting all the data sometimes can sometimes not, the time interval, usually afternoon can't be the next morning, and then can't.

mostafaberg, 2018-01-23:
@JunJun1994 Is that issue happening against one of our example firmwares? I'm sorry it took some time to respond, I'm not sure how to follow with this since you're testing against an actual glucose monitor and not one of our example frimwares, it might be many different reasons causing this and it's outside what we know since the glucose meter might behave in a different way than expected, can you test against our dev kit and verify that you don't see the same issue ?

also, can you try the official app for this device and see if the same issue happens there too ?

mostafaberg, 2018-03-07:
hi @JunJun1994 , have you solved your issue yet ? I'll be closing the issue during this week if there are no more updates available from your side

philips77, 2021-04-30:
> I'll be closing the issue during this week...

@mostafaberg it has been a reeeeeeeeealy long week.

closed 2021-04-30 by philips77

mostafaberg, 2021-04-30:
I was counting Nordic weeks 😂