nordicsemi/IOS-nRF-Toolbox #45: DFU. Operation not permitted

Xiahaiquan, 2018-01-11:
HI 
When I upload the Zip file,  remind Operation not permitted . what's the reason?
Do anyone have the same problem 。

philips77, 2018-01-12:
It depends. You have to give more details. Which SDK are you using, what ZIP file, etc.
Without it we can't help you. There is a lot of reasons you got this issue.

Xiahaiquan, 2018-01-12:
I found the reason. I don't have config DFU  environment. Thank you.
[image]

Xiahaiquan, 2018-01-18:
HI
 when I  used DFUModel ，it is success. after I scan CBPeripheral use the CBCentralManager. But the `**- (void) centralManager:(CBCentralManager*) manager didPeripheralSelected:(CBPeripheral*) peripheral;** `this method in MyViewController it not work.
How can I fix it? Thank you

mostafaberg, 2018-01-23:
Seems like your issue has been resolved, and you have asked a different question in the same thread, will close this one and if you still have another question please open a new issue :)

closed 2018-01-23 by mostafaberg