nordicsemi/IOS-nRF-Toolbox #46: How do I integrate into objects-c projects?

Coder-Yfeng, 2018-01-12:
My Obj-C projects is using Cocoapods and install iOSDFULibrary,but I Can't find the header file?
By the way, your Obj-C demo updateFirmware,the demo will crash;

mostafaberg, 2018-01-12:
Good morning @Coder-Yfeng !
I'm a bit confused, is the issue related to the DFU Library or nRF Toolbox ?
and which objective C demo are you talking about ? all our codebase is in Swift :)

if you clarify a bit more we will be able to help, DFU Library works well with Objective-C and Swift

Coder-Yfeng, 2018-01-12:
//

Coder-Yfeng, 2018-01-12:
nRF-Toolbox obj-c demo,the demo will crash,
https://github.com/zhangferry/nRF-Toolbox
libc++abi.dylib: terminating with uncaught exception of type NSException

Coder-Yfeng, 2018-01-12:
Single bank DFU upload

mostafaberg, 2018-01-12:
@Coder-Yfeng This is not our project so we can't support that, we can only support our own products :)

Coder-Yfeng, 2018-01-12:
ok!thank you

closed 2018-01-23 by mostafaberg