nordicsemi/IOS-nRF-Toolbox #49: Unable to get UART connection on iOS 11

idokleinman, 2018-02-22:
Could not connect to the nRF52840 preview DK eval kit running the ble_peripheral/ble_app_uart on iOS 11, tried on iPhone 7 and 8. Connection does not get established. 
Tried the app on an Android - no issues.

Please advise

mostafaberg, 2018-03-07:
This seems weird indeed, if you try connecting with nRF Connect on iOS, are you able to find the uart TX and RX characteristics ?

rectoyon, 2018-10-03:
i'm experiencing something similar. in ios 12 UART works on my iphone x but not on my ipad. In the ipad it sends a 0x01 0x00 byte array. But, writes never work.

this is only after I added the dfu service to my app.

Has anyone else been working with this combination?

philips77, 2018-10-03:
Isn't it a cashing issue? Did you try to restart Bluetooth before connecting to wipe cache?

closed 2018-10-03 by philips77