nordicsemi/IOS-nRF-Toolbox #50: Can xCode9 compile and run the swift3.0 version of the Nordic SDK?

HelloAnpengyu, 2018-03-07:
Can xCode9 compile and run the swift3.0 version of the Nordic SDK? I am a beginner......
Thanks a lot！

mostafaberg, 2018-03-07:
That would be Swift 3.2 and not 3.0, however, this app is written in Swift 4,  what Nordic SDK are you referring to ?

HelloAnpengyu, 2018-03-07:
I want to work with nRF Toolbox, but I found swift 4.0 has less tutorial than swift 3.0. So I want to downlod previous version nRF Toolbox Sdk with swift 3.0. But xCode9 supports swift4.0 and swift 3.2, I am not sure if the privious version Sdk can work in xcode9.....

mostafaberg, 2018-03-07:
I don't think it's smart to start from Swift 3.0, it'll eventually be deprecated, just start from Swift 4.0, and any tutorial that is written in Swift 3.0 can easily be ported to Swift 4.0, that being said, this repository just holds the nRF Toolbox example application, and it is not an SDK.

Previous versions of the SDK not only upgrade Swift version, but they also include lots of features and improvements/bugfixes, You don't really want to use older versions if you're learning something new.

best of luck :)

closed 2018-03-07 by mostafaberg