nordicsemi/IOS-nRF-Toolbox #54: IphoneX and iphone8 dfu

js9e13, 2018-09-05:
The app will stuck at enabling DFU Bootloader. After a while, it will return unknow error(code 0).
However, the same packet is used and it works on iphone 7 , iphone 6+, iphone 6.

Thank you for investigating this.

juicejams, 2018-10-05:
it goes to enabling bootloader, then Starting DFU then I get the error: Enabling notifications failed.
I'm using iOS 12 and iPhone X .

js9e13, 2018-11-29:
Yeah, I faced the same problem.

philips77, 2018-11-29:
Hi, could you send logs? Are you using the app from AppStore, or compile it in XCode? Logs are only generated in the XCode version.

juicejams, 2018-12-01:
Hi I'm using the app from the app store. Thanks

se7ensong, 2019-01-04:
HI @philips77 ,
Here are the logs. I think it stucks at https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/e44ea97721bc6d995bcd4099606ef04d48688653/nRF%20Toolbox/BluetoothManager/NORBluetoothManager.swift#L84

[DFU_log_iPhoneX&iPhone6.txt](https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/files/2727121/DFU_log_iPhoneX.iPhone6.txt)

Thank you very much.

se7ensong, 2019-01-10:
I have done BLE sniffing and found the problem is that iPhone that supports 2Mbps PHY layer will send PHY update request which is not handled correctly in the firmware. So, there is no problem with this repo. I pushed a fix to adafruit arduino bootloader. https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/40

closed 2019-01-10 by philips77