nordicsemi/IOS-nRF-Toolbox #56: Scanning for the DFU Bootloader ...

HongCongZhi, 2018-10-09:
I used DFU in IOS-nRF-Toolbox and prompted "Scanning for the DFU Bootloader ...", it didn't respond.

hy15816, 2019-08-01:
@philips77 
I found the same problem.
Just like this one: 
[image]

The program stays there any times,and peripheral always connented .Disconnect only when power is cut off。
The DFU Library verstion is 4.1.1 
iOS 12.3   - iPhone 7
xcode 10.1 (10B61)

Can you help me ？ or give me some suggests, thank you!

philips77, 2021-04-30:
Hello, sorry for replying so late.
This issue is related to DFU Library, and an older version of it. It has been fixed with https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/367 and released in version 4.7.1. Please use most current version of the DFU library (which right now is 4.10.1, released today).

closed 2021-04-30 by philips77