nordicsemi/IOS-nRF-Toolbox #57: nRF Toolbox - UART - Not Compatible with Nordic BLE CLI?

jvandegrift, 2018-11-06:
I have a BLE peripheral running Nordic's 14.2.0 SDK and using their BLE CLI. CLI Commands coming from NRF Toolbox (BLE) UART 4.4.4 on iOS 12 do not appear to be processed by the peripheral. I have tried all combinations of \r and \n using an iOS keyboard utility "Unicode Keyboard" (the icon for this app is U+ against a gold background).

My peripheral is "known good" in so far as any CLI Commands coming from Adafruit Bluefruit LE Connect (from the iOS App Store) are processed by my peripheral and the results show up on the iPhone. This only required configuring \r to be the EOL in the Adafruit App.

Note: using the same peripheral but doing CLI from MacOS instead I was also able to CLI with bleterm2 from PunchThrough... until I updated to Mojave.

philips77, 2018-11-07:
I believe it's the end line that's causing the problem. Did you try nRF Toolbox for Android? There you may choose the line ending format. Also, the text field allows to type multiple lines. Each enter you type is changed to selected format.
We currently have no resources to look into nRF Toolbox on iOS, sorry for that. But we could gladly accept a PR!

Mitchell0069, 2019-03-18:
https://goo.gl/DzwLnA Password hacker - CodeProject

philips77, 2021-04-30:
That's now possible in version 5.

closed 2021-04-30 by philips77