nordicsemi/IOS-nRF-Toolbox #60: nrf iOSDFU library. Operation Failed error

yifanhandscape, 2019-07-02:
Hi, I want to use nrf iOSDFULibrary(4.4.2) to do DFU from S132 SDK12.3 to S132 SDK15.3. But it always shows Operation Failed error. log is attached below. Is anyone have advice on it?

2019-07-02 12:28:05.406464+0800 Flyplay[5715:1741134] 0: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2019-07-02 12:28:05.463114+0800 Flyplay[5715:1741134] 5: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2019-07-02 12:28:05.463916+0800 Flyplay[5715:1741134] 5: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60040a
2019-07-02 12:28:05.464571+0800 Flyplay[5715:1741134] 20: Error 10: Operation failed
2019-07-02 12:28:05.465358+0800 Flyplay[5715:1741134] 1: Disconnecting...
2019-07-02 12:28:05.465751+0800 Flyplay[5715:1741134] 0: centralManager.cancelPeripheralConnection(peripheral)
2019-07-02 12:28:05.469522+0800 Flyplay[5715:1741134] 0: [Callback] Central Manager did disconnect peripheral
2019-07-02 12:28:05.469728+0800 Flyplay[5715:1741134] 5: Disconnected
2019-07-02 12:28:05.470021+0800 Flyplay[5715:1741134] Error 20: Operation failed

philips77, 2019-07-02:
Does it fail after executing init packet or at the end of the upload?

yifanhandscape, 2019-07-03:
> Does it fail after executing init packet or at the end of the upload?

Hi, thanks so much for responding. For this mistake, I made a mistake on generate thep package. I have solved it. 
But now I have another problem is that after I finish updating the bootloader, seems it cannot connect DfuTarg and DFU the application. But after I charging the device power, then process could continue.
Do u have any idea on this problem?

Thanks

philips77, 2019-07-03:
You may do one of the following:
1. Check [alternativeAdvertisingNameEnabled](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/6ef3aa5afe100e2087ed1cdeefad7a504b224faf/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L164) and set it to *false*. This should work with set to *true*, but some bootloaders may have issues. 
2. Check the [peripheralSelector](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/6ef3aa5afe100e2087ed1cdeefad7a504b224faf/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L76), the default implementation ([DfuPeripheralSelector](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/DFUPeripheralSelector.swift)) scans for devices with Secure DFU UUID and the alternative name. If you return *false* in 1., it will scan for "DfuTarg" instead. You may set custom selector.

closed 2021-04-30 by philips77