nordicsemi/IOS-nRF-Toolbox #61: Buttonless DFU not working, Error 4: Operation Failed

jthansen, 2019-07-02:
I changed "alternativeAdvertisingNameEnabled" to false like I saw in another thread. But the operation is still failing. 

```
Verbose: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
Verbose: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
Application: Buttonless DFU indications enabled
Warning: Application with buttonless update found
Verbose: Trying setting bootloader name to Dfu58664
Verbose: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
Debug: peripheral.writeValue(0x0a000000000000000208, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Info: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
Info: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):030000
Error: Error 4: Operation failed
Verbose: Disconnecting...
Debug: centralManager.cancelPeripheralConnection(peripheral)
Debug: [Callback] Central Manager did disconnect peripheral
Info: Disconnected
```
nRF Toolbox doesn't work but nRF Connect does. Why would that be?

philips77, 2019-07-02:
From `Verbose: Trying setting bootloader name to Dfu58664` I conclude that you didn't disable the alternative adv name

jthansen, 2019-07-02:
I cleaned my build folder and then it worked. Thanks!

closed 2019-07-02 by jthansen