nordicsemi/IOS-nRF-Toolbox #64: Not disconnecting device in UART

sandhu0409, 2019-07-12:
First time it shows all BLE devices. when I connected with one device and sending data works perfectly. But when I will disconnect and try to reconnect it didn't show in scanning peripheral list even other devices not able to find that ble device.

philips77, 2019-07-13:
How about killing the nRF Toolbox app? Or restarting the phone/Bluetooth?
Seems like the device is still connected after you click disconnect, for some reason.

sandhu0409, 2019-07-13:
Yeah after restarting bluetooth it works fine. But after reconnect issue remains the same.

philips77, 2019-07-13:
Looks like the app, or some other app, is still using this connection after uart disconnected. Calling cancelConnection() method disconnects only the virtual connection, but if some other is still open, the iOS will keep the physical Bluetooth link open. To be sure that the device disconnects you should implement disconnect feature on your peripheral side, e.g. after sending some op code.

sandhu0409, 2019-07-13:
I have disconnected it but still its not working

sandhu0409, 2019-07-15:
I think Apple not disconnected main Ble connection and stores in Cache. Don't know its Apple bug or something else

philips77, 2021-04-30:
I'm closing the issue as it was related to older version of the app.

closed 2021-04-30 by philips77