nordicsemi/IOS-nRF-Toolbox #75: Secure DFU: "Error 3: Writing is not permitted."

rogmabi, 2019-09-11:
Hi,

We have tested our DFU distribution package with the Android nRF-Toolbox app and it worked fine. However on iOS, we are getting the following error:

**`Error 3: Writing is not permitted.`**

Here is the full Xcode console logs:

```
D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to StrictIDAFAA...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to StrictIDAFAA
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Secure DFU...
I: Connected to StrictIDAFAA
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
V: Trying setting bootloader name to Dfu84741
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753834373431, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
I: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
A: Response (Op Code = 2, Status = 1) received
A: Bootloader name changed successfully
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
I: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
A: Response (Op Code = 1, Status = 1) received
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Scanning for the DFU Bootloader...
I: DFU Bootloader found with name Dfu84741
V: Connecting to Dfu26555...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to Dfu26555
V: Discovering services...
D: peripheral.discoverServices([FE59])
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
V: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
A: Secure DFU Control Point notifications enabled
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
E: Writing to characteristic failed. Check if Service Changed service is enabled.
E: Error 3: Writing is not permitted.
V: Disconnecting...
D: centralManager.cancelPeripheralConnection(peripheral)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected
```

Do you have some troubleshooting steps or recommendations to work around this issue?

Thanks :)

rogmabi, 2019-09-13:
Hi @philips77, looks like you are correct, restarting Bluetooth worked. And yes, we are already in production, so for this one-time, we will instruct our early customers to either use an Android device or restart Bluetooth before attempting the OTA DFU. Once we've done the major SD upgrade, I guess we can add the SC service for future DFUs...

philips77, 2019-09-13:
If you don't have it now, you should not add it later. That would require updating the bootloader using the old bootloader, which services would be cached therefore making the second update not possible without restarting Bluetooth.
We had a case, where users were asked to use Android to update, as it was 100% not possible to update the device using iOS. Their device didn't have a button to switch to bootloader, which would help. Removing SC characteristic is easy, adding it is challenging. You doing need it in bootloader, the services won't change and no additional features are needed.

closed 2021-04-30 by philips77