nordicsemi/IOS-nRF-Toolbox #78: For multiple user device

pandapancake, 2019-11-07:
Does nRF-Toolbox support multiple user device? I have a blood pressure monitor that can have up to 4 different users (Fora P20b). But every time it connected only connected to user 1. I can't find the code/func for checking the users. This is my first time coding bluetooth device so I'm not sure where to look. I put a bunch of breaks but still can't find it. Thanks!

```import UIKit
import CoreBluetooth

class BloodPressureMonitorVC: BaseViewController, CBCentralManagerDelegate, CBPeripheralDelegate, ScannerDelegate {
    
    //MARK: - ViewController Properties
    var bpmServiceUUID                                : CBUUID
    var bpmBloodPressureMeasurementCharacteristicUUID : CBUUID
    var bpmIntermediateCuffPressureCharacteristicUUID : CBUUID
    var batteryServiceUUID                            : CBUUID
    var batteryLevelCharacteristicUUID                : CBUUID
    var bluetoothManager                              : CBCentralManager?
    var connectedPeripheral                           : CBPeripheral?
    var systolicReading: String?
    var diastolicReading: String?
    var heartRateReading: String?
    var time: Date?
    var hasEHR = true
    
    //MARK: - Referencing Outlets
    @IBOutlet weak var deviceName: UILabel!
    @IBOutlet weak var battery: UIButton!
    @IBOutlet weak var connectionButton: UIButton!
    @IBOutlet weak var systolic: UILabel!
    @IBOutlet weak var systolicUnit: UILabel!
    @IBOutlet weak var diastolic: UILabel!
    @IBOutlet weak var diastolicUnit: UILabel!
    @IBOutlet weak var meanAp: UILabel!
    @IBOutlet weak var meanApUnit: UILabel!
    @IBOutlet weak var pulse: UILabel!
    @IBOutlet weak var timestamp: UILabel!
    @IBOutlet weak var saveButton: UIButton!
    
    //MARK: - Referencing Actions
    @IBAction func connectionButtonTapped(_ sender: AnyObject) {
        if connectedPeripheral != nil {
            bluetoothManager?.cancelPeripheralConnection(connectedPeripheral!)
        }
    }

    //MARK: - UIViewController methods
    required init?(coder aDecoder: NSCoder) {
        bpmServiceUUID                                = CBUUID(string: ServiceIdentifiers.bpmServiceUUIDString)
        bpmBloodPressureMeasurementCharacteristicUUID = CBUUID(string: ServiceIdentifiers.bpmBloodPressureMeasurementCharacteristicUUIDString)
        bpmIntermediateCuffPressureCharacteristicUUID = CBUUID(string: ServiceIdentifiers.bpmIntermediateCuffPressureCharacteristicUUIDString)
        batteryServiceUUID                            = CBUUID(string: ServiceIdentifiers.batteryServiceUUIDString)
        batteryLevelCharacteristicUUID                = CBUUID(string: ServiceIdentifiers.batteryLevelCharacteristicUUIDString)
        super.init(coder: aDecoder)
    }
    
    override func viewDidLoad() {
        super.viewDidLoad()
        serviceAuthForSIPPA(authorization: "XhLI05lcrv")
        
        if hasEHR == true {
            connectionButton.isEnabled = true
        } else {
            connectionButton.isEnabled = false
            connectionButton.backgroundColor = .lightGray
        }
    }
         
    override func viewDidAppear(_ animated: Bool) {
        guard verifyFileIsValidXML(fileName: ccdFileName) != nil else {
            connectionButton.isEnabled = false
            connectionButton.backgroundColor = .lightGray
            return
        }
[... 234 lines cut ...]
            }
        }
    }

    //MARK: - Segue handling
    override func shouldPerformSegue(withIdentifier identifier: String, sender: Any?) -> Bool {
        return identifier != "scan" || connectedPeripheral == nil
    }
    
    override func prepare(for segue: UIStoryboardSegue, sender: Any?) {
        if segue.identifier == "scan" {
            let nc = segue.destination as! UINavigationController
            let controller = nc.children.first as! ScannerViewController
            controller.filterUUID = bpmServiceUUID
            controller.delegate = self
        } else if segue.identifier == "bloodPressureMonitorData" {
            let controller = segue.destination as! AddBloodPressureMonitorDataVC
            controller.systolicReading = systolicReading
            controller.diastolicReading = diastolicReading
            controller.heartRateReading = heartRateReading
            controller.timestamp = time
        }
    }
}
```

philips77, 2019-11-08:
As far as I remember, the user selection is done on the device. The device we have has buttons to select a user before starting measurement. The user ID is then sent to the iPhone. @NickKibish you did this profile recently, can you confirm? Do you display the user ID in the new app version?
Btw, @pandapancake  please check the branch with new version.

pandapancake, 2019-11-08:
@philips77 Thanks! Because we have Android app too and they have it in the UI for choosing user number. So I thought it is somewhere in the framework. :)

closed 2019-11-08 by pandapancake