nordicsemi/IOS-nRF-Toolbox #94: DFU: unable to use iOS to do OTA, failed inconsistently.

parapentiste, 2020-03-31:
Hi, 

How are you?
We have encountered issues on iOS DFU. We have tested our DFU distribution file with both iOS version and Android version. The Android version works fine. But the iOS version failed each time. I have read through the issues that other people encountered here, but none are the same as what we encountered.

After testing it over 10 times, there are roughly 3 different failure scenario.
A couple of times, it showed "I: Disconnected by the remote device", and a couple of times it stuck in "V: Scanning for the DFU Bootloader...". And one time, it showed " E: Error 7: SD version check failed", I was unable to reproduce the last scenario. 

Here are the Xcode logs for the 3 different scenario :

1st log:

> D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to TT02...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to TT02
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Secure DFU...
I: Connected to TT02
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device

2nd log:

> D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to TT02...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to TT02
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Secure DFU...
I: Connected to TT02
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
V: Trying setting bootloader name to Dfu79256
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753739323536, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Scanning for the DFU Bootloader...

3nd log:

> D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to TT02...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to TT02
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Secure DFU...
I: Connected to TT02
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
V: Trying setting bootloader name to Dfu32139
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753332313339, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Scanning for the DFU Bootloader...
I: DFU Bootloader found with name DfuTarg
V: Connecting to DfuTarg...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to DfuTarg
V: Discovering services...
D: peripheral.discoverServices([FE59])
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
V: Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
A: Secure DFU Control Point notifications enabled
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
A: Command object info (Max size = 256, Offset = 0, CRC = 00000000) received
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x01018e000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
A: Packet Receipt Notif disabled (Op Code = 2, Value = 0)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x128b010a450801124108ff0110341a02ca012000, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x2800300038f898074224080312208acb4c8d6c8e, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0xf0bab4ade41ab72e1bf030b3b995dcfbdf403044, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x765f0d9af650480052040801120010001a405eab, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x8b00b429f6a08ee0d7922fbb8151250c6efe120d, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x614ac7283e465fb4707cc30897d923a9cb7acdb3, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x7476da2c1de637624a38c68be3e3ca0ff7795951, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
V: Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x9f18, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
A: Command object sent (CRC = E3606AC4)
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 6003018e000000c46a60e3
A: Checksum (Offset = 142, CRC = E3606AC4) received
V: Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
I: Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
I: Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60040b07
E: Error 7: SD version check failed
V: Disconnecting...
D: centralManager.cancelPeripheralConnection(peripheral)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected

I was not able to reproduce the last one, but the 1st one appeared a lot.

The bootloader is based on SDK v.16.0. Tests were done on iOS13.2.3.
 
I wonder do you have any suggestions regarding what we can do with iOS app or firmware regarding this issue? 

Thank you.

philips77, 2020-03-31:
Hi,
Are you testing using nRF Toolbox, the current version on AppStore, or using your app that includes the [DFU Library](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library)?
The current version of nRF Toolbox hasn't been updated for a while, and there were changes related to performance on iOS in the latest lib versions. You ay try nRF Connect for iOS (available on AppStore), which should have the latest version.

Regarding the "SD version check failed" error. This could happen if you try to update SD+BL+App twice, or with an invalid file. Each distribution package has "sd-req" parameter, which specifies which SoftDevices it's compatible with. If you updated your device with a new SoftDevice, and then try to upload it again, it will fail, as the new SD has new SD ID, and does not match the old one. You need to restore the initial test setup to make it work again, that is flash the original firmware that you update from.

Regarding
> V: Trying setting bootloader name to Dfu79256
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753739323536, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device

This looks a bit suspicious. Usually, the device should reply with 0x10-02-01 (response-for set name command-success), after which the lib should send 0x01 to reset the device. Yours has disconnected prior to that, so the renaming feature didn't work. This **seems like a bug in the library**, however the root cause is on your device. It should reply as mentioned above. But when it fails, the library fails to behave correctly. I'll look into it.

philips77, 2020-03-31:
The first issue, with "DFU service not found" most probably it related to cacheing. Try disabling Bluetooth on your phone and enabling it again to clear the caches, and try again. It happens when you connect to a device that during previous connection had different services and didn't have Service Changed characteristic. In that case iOS had cached the services and doesn't let them go easily :)

parapentiste, 2020-04-01:
hi Philips, 

Thank you so much for your swift reply. Answering your question, I am testing using nRF toolbox from this github repository.
On your suggestion, I have also tested it with the newest version of nRF Connect for iOS from AppStore. It failed to do OTA as well.
With nRF toolbox, I was only able to reproduce the first failure scenario for the pass two days. Each time it failed, the logs was the same as the first one. I tried to turn Bluetooth off and on as you suggested. But the result was still the same. 
Do you have any other suggestion regarding the first one?

Thank you.

philips77, 2020-04-01:
Sorry, in the first log you have "Secure DFU Service found", not "not found":
> V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device

This may be an issue with the device. I'm not sure why can it fail at this moment. You need to fo some sniffing to find out the root cause.

aLHW1, 2020-04-05:
D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to TMC_074ec887a6c1...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to TMC_074ec887a6c1
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Secure DFU...
I: Connected to TMC_074ec887a6c1
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
V: Trying setting bootloader name to Dfu10227
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753130323237, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Scanning for the DFU Bootloader...

DFU upgrade process to this is no longer conducted

philips77, 2020-04-05:
This is related to this issue: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/365
I'll try to fix it on Monday.

aLHW1, 2020-04-08:
I: Services discovered
V: Secure DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: FE59)
I: DFU characteristics discovered
V: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
V: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
A: Buttonless DFU indications enabled
W: Application with buttonless update found
V: Trying setting bootloader name to Dfu97213
V: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
D: peripheral.writeValue(0x02084466753937323133, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
E: Buttonless service not configured, see: https://devzone.nordicsemi.com/f/nordic-q-a/59881/advertising-rename-feature-not-working/243566#243566. To workaround, disable alternative advertising name.
V: Connecting to TMC_074ec887a6c1...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral

Since the update to 4.7.1, the connection to bluetooth has been looped

philips77, 2020-04-08:
Yes. And you have a solution now in the log:

> E: Buttonless service not configured, see: https://devzone.nordicsemi.com/f/nordic-q-a/59881/advertising-rename-feature-not-working/243566#243566. To workaround, disable alternative advertising name.

philips77, 2020-04-08:
Well.. In 4.7.1 it wasn't me intention to make it go in a loop, but adding this message. Before, the lib was assuming the device succeeded to reboot into DFU mode and trying to connect to it. I agree that it would be better to stop DFU if this issue is discovered, or turn the alternative name off, but then you wouldn't know you have an issue in your app.

closed 2020-04-09 by philips77