nordicsemi/IOS-nRF-Toolbox #103: "[CoreBluetooth] XPC connection invalid" error

Sikandarkhan, 2020-09-15:
Greetings,
Once the app has launched, I'm getting an error in the Xcode console that says: "[CoreBluetooth] XPC connection invalid"

Below is what I'm using hardware/software wise:

macOS Catalina v10.15.4
Xcode v11.5
iPhone 6 -- iOS v12.3

I also made sure to add the NSBluetoothAlwaysUsageDescription and NSBluetoothPeripheralUsageDescription permissions to the info.plist file.

philips77, 2020-09-15:
This is just a warning. It happens when a `CBCentralManager` instance is disposed. In the app and in our libraries we often use one manager instance per profile to separate functionality and allow multiple delegates. iOS would prefer a singleton which lives as long as the app, but everything works as expected with multiple instances.

Sikandarkhan, 2020-09-19:
After the update, I am not getting values for systolic and diastolic. Only seeing battery %

[image]

please suggest me on this.

Sikandarkhan, 2020-09-19:
I realised the issue is with Bluetooth pairing process. After unpairing and pairing to Bluetooth this code started working appropriately. 

I was spending days and days to debug code and writing queries to firmware developers.

Glad that this simple fix resolved my query. 
[image]

Hope this answer will be helpful to others in future.

closed 2020-09-19 by Sikandarkhan