nordicsemi/IOS-nRF-Toolbox #104: UART App Not Sending EOL

dbeckwith01, 2020-10-21:
I'm using nRF Toolbox v5.0.2 with the Nordic BLE CLI on a DK.  When I create and send a text command the command is sent (confirmed on the logs) but the EOL doesn't seem to be sent to execute the command on the CLI.

I have gotten around this by creating a second command to send a hex 0x0D down to the device.

philips77, 2020-10-23:
You need to type a new line at the end of the first command. The eol selection defined to what values the new lines are encoded.

dbeckwith01, 2020-10-26:
@philips77 Can you please provide a little more detail.  I tried adding a new line at the end of the command and when I come back the new line is no longer in the command window.

philips77, 2020-10-26:
@NickKibish ?

closed 2023-10-09 by NickKibish