nordicsemi/IOS-nRF-Toolbox #112: Question: BLE Device simulator and/or compatible DevKit

Dario-GoldenSpear, 2021-05-17:
Hi, 

My name is Dario. I am located in Argentina and currently working as an iOS dev. on a project based on nRF5x prototype devices for a company located in Spain.

I was able to compile & run this project and as a next step I'd like to be able to interact, through BLE, with a nrF5x device (or a simulator).

QUESTIONS:
1) Do you have, (or can I get from another source) a BLE Device simulator that I can use to connect to an iPhone and allows me to interact from the the iOS-nRF-Toolbox app (until I get access to the real prototype hardware)?
I found this: https://github.com/NordicSemiconductor/asset-tracker-cloud-device-ui-server-js, but from the documentation I was not able to identify if the simulated devices (Cat Tracker?) serves our purpose (nrF5x simulated device).

2) Is it possible to order and receive a devKit or another nrF5x based device from Argentina? I was considering the Thingy:52 (https://www.avnet.com/shop/us/products/nordic-semiconductor/nrf6936-3074457345634992668?CMP=EMA_OEMSecrets_inventoryfeed_VSE).

Thanks in advance.

philips77, 2021-05-20:
Hello,

For emulating your BLE hardware before it's ready you may use nRF Connect for Mobile. We have it for iOS and Android, of which the Android version has more features that you may find useful.
You will need to create GATT Server Configuration with services that your device will have in the future, create advertising packet similar to one that your device will advertise with and set Connectable flag in it. You should see nRF Connect advertising in your iOS app, and after connection a new tab will open in nRF Connect, from where you may send notifications or set values for different readable characteristics. You may also find Macros feature useful if you wish to automate it.

Dario-GoldenSpear, 2021-05-20:
Sounds cool!, thanks @philips77 for the suggestion 👌

closed 2021-05-20 by Dario-GoldenSpear