nordicsemi/IOS-nRF-Toolbox #114: Incorrect unit for blood pressure

Walter0697, 2021-08-05:
It seems like for the official GATT specification, bit 0 will have a value of 0 if the unit is mmHg, and a value of 1 if the unit is kPa. However, the logic in the code seems to be reserved. 

```
private extension Flag {
    static let unitFlag: Flag = 0x01
    static let timeStamp: Flag = 0x02
    static let pulseRate: Flag = 0x04
}

struct BloodPressureCharacteristic {
    
    let systolicPressure: Measurement<UnitPressure>
    let diastolicPressure: Measurement<UnitPressure>
    let meanArterialPressure: Measurement<UnitPressure>
    let date: Date?
    let pulseRate: Int?
    
    init(data: Data) throws {
        let flags: UInt8 = try data.read()
        let unit: UnitPressure = Flag.isAvailable(bits: flags, flag: .unitFlag) ? .millimetersOfMercury : .kilopascals
```

The incorrect logic cant be found in `BloodPressureCharacteristic.swift` and `BloodPressureCharacteristic.swift` 

If I am correct, the correct code should be 
`let unit: UnitPressure = Flag.isAvailable(bits: flags, flag: .unitFlag) ? .kilopascals : .millimetersOfMercury `

philips77, 2021-08-05:
On the other hand, after testing the whole app, you have found only one bit wrong! Look how many bits we got right!

NickKibish, 2021-08-09:
Ok, I'll check that