nordicsemi/IOS-nRF-Toolbox #122: MCUManager vs SwiftCBOR target difference build error

Dario-GoldenSpear, 2022-02-13:
Hi,

I am trying to build the latest version from the master branch: https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/commit/53605f540112b1ca9b2820bdb752039d3c16f4b3

(same problem if I try the latest release: https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/tree/5.0.7)

Xcode version: 13.2.1

After running `pod install` and updating the targets versions to 12.0 as suggested by Xcode:
[image]

When trying to build I get the following error:
```
Build target McuManager with configuration Debug
error: The package product 'SwiftCBOR' requires minimum platform version 10.0 for the iOS platform, but this target supports 9.0 (in target 'McuManager' from project 'McuManager')
```

To the best of my knowledge the problem is that SwiftCBOR's project has been updated 2 months ago, setting the package's min. target to 10.0:
https://github.com/unrelentingtech/SwiftCBOR/blob/master/Package.swift

while McuManager's package is still set to 9.0 (develop branch):
https://github.com/NickKibish/IOS-nRF-Connect-Device-Manager/blob/develop/Package.swift

Also in the master branch:
https://github.com/NickKibish/IOS-nRF-Connect-Device-Manager/blob/master/Package.swift

dinesharjani, 2022-02-14:
Hi @Dario-GoldenSpear - thanks for making us aware of this.

In the Device Manager library we targeted an earlier version of CBOR that doesn't require the higher minimum platform version.

dinesharjani, 2022-02-14:
@Dario-GoldenSpear making a fix. Give me a few minutes.

dinesharjani, 2022-02-14:
@Dario-GoldenSpear I think we fixed the issue. Let us know.

Dario-GoldenSpear, 2022-02-14:
@dinesharjani yes, this fix solved the issue, thanks!.

closed 2022-02-16 by dinesharjani