nordicsemi/IOS-nRF-Toolbox #125: Missing package product 'McuManager package'

Terencewyt, 2022-06-27:
Hi Sir,

I am looking to use the BLE UART features with my nRF development board, I download the **IOS-nRF-Toolbox** and try to build and run, but it said "Missing package product '**McuManager** package', and when i check my Package Dependencies, it contains **IOSMcuManagerLibrary**, is it the same?
and I did try to add package from "**In Xcode, go to File → Swift Packages → Add Package Dependency... and add https://github.com/JuulLabs-OSS/mcumgr-ios.git**", after that there is another error where SwiftCBOR version is conflict

I am relatively new to the IOS development, would like you give some help , my first target is to build successfully the code and run in my iphone. thanks a lot in advance

Terencewyt

dinesharjani, 2022-06-29:
Thanks for letting us know.

We're working on a fix: https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/pull/126

Also, the correct SPM for McuMgr is the following: https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager

Terencewyt, 2022-07-06:
Thanks dinesharjani
After use the latest Code, the "McuMgr" problem is fixed, but there are quite a lot other problem, and I still cannot build successful
1) many files have similar error of the following:
**Could not read serialized diagnostics file: error("Invalid diagnostics signature")
Command CompileSwift failed with a nonzero exit code**

2) Quite a lot of problem for DFUFirmware
because I only want to try the BLE UART features, anyway to disable the DFU feature?

attached is the print screen shot
[image]

Thanks in advance :)

dinesharjani, 2022-07-07:
This is weird. I'll try to have a look today.

Terencewyt, 2022-07-11:
> This is weird. I'll try to have a look today.

thanks your help

dinesharjani, 2022-07-11:
Did you check out the master branch ?

Terencewyt, 2022-07-11:
> Did you check out the master branch ?

I download the code from "Code -> Download ZIP" (https://github.com/NordicSemiconductor/IOS-nRF-Toolbox.git)

Terencewyt, 2022-07-13:
or any Xcode version can build the source successful? I try to fix for long time, but cannot get it. 
so many "Command CompileSwift failed with a nonzero exit code" and many other compile error for DFU

Terencewyt, 2022-07-13:
I can build successful now, 
1)  add a User defined settings "swift_enable_batch_mode" to NO
2) fix many programming error "e.g use "try" and "catch" to avoid error of" **Call can throw, but it is not marked with 'try' and the error is not handled"**

Terencewyt, 2022-07-13:
> @Terencewyt try now when you can

i will try it, thanks your help

dinesharjani, 2022-07-13:
> I can build successful now,
> 
>     1. add a User defined settings "swift_enable_batch_mode" to NO
> 
>     2. fix many programming error "e.g use "try" and "catch" to avoid error of" **Call can throw, but it is not marked with 'try' and the error is not handled"**

I fixed issue 2 on master. It should build without issues now.

Terencewyt, 2022-07-13:
> I fixed issue 2 on master. It should build without issues now.

YES, i build without problem, thanks again, the issue can be closed

closed 2022-07-13 by Terencewyt